eerimoq/simba #164: DHT driver only supports AVR

simonlbn, 2017-10-28:
Hey,

I wanted to try out the DHT driver for some of the DHT11 / DHT21 sensors I have, but it's AVR only and I can see it depends on time_micros() which doesn't seem implemented e.g. on ESP8266. So, my main question is: Why was it changed to use time_micros in commit 
4c2b432c80143c356a2bdcce25e2c9c9f5fec19c ?

If it's just more efficient, could the "old" way be used for MCU's which does not have time_micros  ?

Separate question is - DHT11 and DHT21 are different (at least in the Arduino driver you need to select the right one as you otherwise get bogus results), so which one(s) does the Simba DHT driver support? Don't have an AVR board handy right now so I can't see what works which was my original plan.

eerimoq, 2017-11-04:
The DHT driver has not yet been tested using real hardware. I have two AOSONG AM2302 (I believe those are DHT22) on my desk, but have not managed to test the driver with them yet.

Using `sys_uptime_isr()` should work, given that its resolution is sufficient. Most ports only increment the uptime every system tick, which usually is in the millisecond range.

The idea of time_micros-family of functions is to have a high resolution time with minimal overhead, to be used in drivers with hard timing requirements.

simonlbn, 2017-11-05:
OK, thanks for the information!

I have a number of DHT21 and DHT11 around which I got before I found out how annoying the protocol is compared to I2C based sensors :-).

I will bring an AVR board on my next trip and may try playing with the driver against real hardware.

I think it makes sense to keep this issue open, so it's simpler to find the state of the driver and your ideas for making it work on more platforms. The DHT sensors are fairly popular / available so other people may care.

PS. extra naming fun / confusion, I recently found DHT12 which seems to be made by AOSONG, but is actually I2C based.

shacal, 2017-11-05:
DHT12 is sensor (with protocol), there is many variants that outputs I2C, SPI and plain Serial packages of it.
Think like ESP ... packed as wimo, nosemcu... heart is ESP, and package is pins, power supply etc.

So, ASONG is package for DHT and exposes I2C.

eerimoq, 2017-11-05:
Great, any help testing/using drivers is appreciated. I started working on the DHT driver 3-4 months ago and it is not yet part of any release. Would be lovely to have it working in the next release =)

Yeah, let's keep the issue open. Ideally the driver documentation should be updated with its current state/limitations.

toomyem, 2018-06-28:
What about system_get_time() from Espressif's sdk? Can't we use it in time_micros() implementation for esp?
```
/**
  * @brief  Get system time, unit: microsecond.
  *
  * @param  null
  *
  * @return System time, unit: microsecond.
  */
uint32 system_get_time(void);
```

toomyem, 2018-06-28:
Ok, I found the problem. Reading bits was wrong. The resulting byte was shifted to much. I've pushed the fix to my branch also.
Now the readings are ok, but as @simonlbn mentioned earlier, the decoding formulas for DHT11 and DHT22 are different. And separate function for both sensors seems to be good idea.
I will try to add this change to my branch later.

eerimoq, 2018-06-28:
Using the Espressif SDK function `system_get_time()` might work as long as it does not enable interrupts. The DHT `read_isr()` functions is called with interrupts disabled, and they should preferably not be enabled within that function. I think the behavior is undefined if so.

As written earlier in this issue interrupts are disabled for a long time reading from the DHT sensor due to hard real time requirements on the bit-banged communication. Since a CRC is provided by the sensor, one could consider have interrupts enabled and return a failure code if the CRC does not match the read data.

Please create a PR with the DHT driver data shift fix. If the `time_micros()` implementation with `system_get_time()` proves to be working we can possibly merge that as well. The better solution would be to implement it using the hardware timer register(s) used by the Simba system tick.

toomyem, 2018-06-28:
Ok, I will create two separate PRs. One with data shift fix, the other with system_get_time() function. My testing showed that this is working, however I did get crc errors from time to time. So I leave the dicision of what to merge to you. 
Regarding hardware based timing, I'm afraid it is not something I can easily handle by myself, but of course I will give it a try.

closed 2018-07-02 by eerimoq