eerimoq/simba #166: [Question] Redundant library/driver for accessing network, wifi/eth!

rahuldeo2047, 2017-11-11:
Can we use [inet.network_interface](http://simba-os.readthedocs.io/en/latest/library-reference/inet/network_interface.html) and / or [inet.socket](http://simba-os.readthedocs.io/en/latest/library-reference/inet/socket.html) instead of [network.esp_wifi](http://simba-os.readthedocs.io/en/latest/library-reference/drivers/network/esp_wifi.html) for ESP8266? So that the same code can be used on ESP32 / any future soc with wifi. Any relevant info/plan as my understanding can be wrong?

shacal, 2017-11-12:
This is currently one blocker for me too to use Simba, as I need to use AVR, ESPs and ST chips for creating network of devices and power usage is one key aspect.
Currently many things are not ”cross” ported, like timing and network. But, for network socket and interface are pretty straight forward to make.
Wifi has its own quirks, see linux and wifi handling :)
I kinda tried to do usage interface for network and wifi (wpa) from linux kernel namings and params, but that was an experiment and needs ton of debugging on every platform, and AVR would be slowest to debug.
I hope that we could come up with some kind of interfaces to use network, wired and wireless. And also come up with provider interface for network chips. But as allways, Simba needs more developers to contribute and my time line for my chip network kinda starts to sift me towarda freertos, and I hate that structure and legacy crap... and namings... uugh. ✌🏻

Simba is clean and namings are simple and it does not hide things :)

simonlbn, 2017-11-15:
On ESP8266, network_interface_wifi works as an abstraction layer, or am I missing something? You do need network.esp_wifi for setting it up as an access point though.

From what I recall network_interface_wifi is what's used if you set the CONFIG_START_NETWORK_INTERFACE_WIFI_SSID and CONFIG_START_NETWORK_INTERFACE.

Or am I missing something?

rahuldeo2047, 2017-11-18:
Hi @simonlbn,
As per my understanding, either of inet.network_interface or network.esp_wifi can be used for an access point in case of ESP8266. 
Hence a **doubt**, whether network.esp_wifi has to be used for esps or inet.network_interface will be sufficient for esps in order to keep the application generic?

I am guessing inet.network_interface will be sufficient for any controller as well as esp. I will try it as and when time permits. Hence the question, what forced us to (or why) keep redundant interfaces for esp?

Will change the question title accordingly.

@shacal: Thank you for putting your thoughts here.

eerimoq, 2017-11-19:
The network_interface module is an attempt to make networking easy to use across all ports. There are certainly improvements to make to reach these goals.

The underlying drivers can be used separately if one prefers.

Initially I planned to create a network manager module (possible part of the network interface module) that handles network connections; re-connecting when needed and so on...

closed 2018-02-03 by eerimoq