eerimoq/simba #168: snprintf and mbedtls_snprintf error in hello_world example with nodemcu board

talkingbeast, 2017-12-06:
hi,i am new to simba,and i follow the tutorial in https://simba-os.readthedocs.io/en/latest/getting-started.html  and try helloworld example. i compiled the code in nano32 board and linux ,it works fine.but when i try to compile this example with board nodemcu,the Terminal show this kind of error message : 
/home/talkingbeast/simba2/simba/simba/3pp/mbedtls/include/mbedtls/platform.h:210:28: error: incompatible implicit declaration of built-in function 'snprintf' [-Werror]
 #define mbedtls_snprintf   snprintf
                            ^
/home/talkingbeast/simba2/simba/simba/3pp/compat/mbedtls/library/mbedtls_error.c:179:13: note: in expansion of macro 'mbedtls_snprintf'
             mbedtls_snprintf( buf, buflen, "CIPHER - Failed to allocate memory" );
             ^
/home/talkingbeast/simba2/simba/simba/3pp/mbedtls/include/mbedtls/platform.h:210:28: error: incompatible implicit declaration of built-in function 'snprintf' [-Werror]
 #define mbedtls_snprintf   snprintf
                            ^
/home/talkingbeast/simba2/simba/simba/3pp/compat/mbedtls/library/mbedtls_error.c:181:13: note: in expansion of macro 'mbedtls_snprintf'
             mbedtls_snprintf( buf, buflen, "CIPHER - Input data contains invalid padding and is rejected" );
.......
/home/talkingbeast/simba2/simba/simba/3pp/mbedtls/include/mbedtls/platform.h:210:28: error: incompatible implicit declaration of built-in function 'snprintf' [-Werror]
 #define mbedtls_snprintf   snprintf
                            ^
/home/talkingbeast/simba2/simba/simba/3pp/compat/mbedtls/library/mbedtls_error.c:687:5: note: in expansion of macro 'mbedtls_snprintf'
     mbedtls_snprintf( buf, buflen, "UNKNOWN ERROR CODE (%04X)", use_ret );
     ^
cc1: all warnings being treated as errors
/home/talkingbeast/simba2/simba/simba/make/app.mk:227: recipe for target 'build/nodemcu/obj/home/talkingbeast/simba2/simba/simba/3pp/compat/mbedtls/library/mbedtls_error.o' failed
make[1]: *** [build/nodemcu/obj/home/talkingbeast/simba2/simba/simba/3pp/compat/mbedtls/library/mbedtls_error.o] Error 1
/home/talkingbeast/simba2/simba/simba/make/app.mk:111: recipe for target 'all' failed
make: *** [all] Error 2
anyone know how to solve this? thanks

simonlbn, 2017-12-07:
Can you include the exact command you run? That makes it simpler to reproduce.

talkingbeast, 2017-12-07:
sorry.the command i run is 

> make -s BOARD=nodemcu
the example is hello_world example.

talkingbeast, 2017-12-07:
the whole compile message is very long,i attach it as i txt file for your consideration.
[helloworld example with nodemcu compile error log.txt](https://github.com/eerimoq/simba/files/1538449/helloworld.example.with.nodemcu.compile.error.log.txt)

talkingbeast, 2017-12-07:
i compile the same example in board linux and nano32 ,it just works fine ,i also attach the log message:
[helloworld example with nano32 and linux board compile message log.txt](https://github.com/eerimoq/simba/files/1538466/helloworld.example.with.nano32.and.linux.board.compile.message.log.txt)

talkingbeast, 2017-12-26:
help,please?