eerimoq/simba #169: Helloworld and blink example don't works with nodemcu

talkingbeast, 2017-12-27:
hello,everyone ,i tried  hello_world example with nodemcu,i don't work.
the attachment is the log file,please help.
[helloword example don't work with nodemcu--console log.txt](https://github.com/eerimoq/simba/files/1589253/helloword.example.don.t.work.with.nodemcu--console.log.txt)

eerimoq, 2017-12-27:
Hello,

it works for me. Compilation and linking without any errors.

Maybe the xtensa toolchains differs? I built mine September 2016.

```
erik@erik-GR8 ~/workspace/simba/examples/hello_world 001 master 
> xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-55-gecfc19a) 4.8.5
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

erik@erik-GR8 ~/workspace/simba/examples/hello_world 001 master 
> 
```

talkingbeast, 2018-01-07:
hi,i try to check the toolchain version,it show this info:

`blink git:master ❯ xtensa-lx106-elf-gcc --version                         ⏎ ✹ ✭
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

blink git:master ❯ `
but i have follow the instruction in [https://simba-s.readthedocs.io/en/latest/getting-started.html](url),the toolchain should be the latest,no the old version 4.8.2,is there something i was mssing??how to setup my simba toolchain to the latest version.
BTW,i have  installed the esp8266 toolchain   for other project before and the required toolchain version is 4.8.2.So that means my machine install more than one toolchain.How to solve this toolchain problem?thanks for you reply.
@eerimoq

eerimoq, 2018-01-07:
Try to prepend the 4.8.5 gcc bin folder to the PATH environment variable.

talkingbeast, 2018-01-18:
i install the simba in another computer followed the instruction in 
https://simba-os.readthedocs.io/en/latest/getting-started.html
the problem is when i compile the blink example,the terminal show this info:

> ➜  blink git:(master) ✗ sudo make -s BOARD=nano32 SERIAL_PORT=/dev/ttyUSB0 upload
Makefile:39: /make/app.mk: No such file or directory
make: *** No rule to make target '/make/app.mk'.  Stop.
is there something i setup wrong?

eerimoq, 2018-01-18:
Run `source setup.sh` in the root folder of the simba repository before building.

closed 2018-02-03 by eerimoq