eerimoq/simba #170: Calling socket_connect repeatedly crashes

ChriThor, 2018-01-03:
NodeMcu 0.9, PlatformIO Espressif 8266 1.5, MS Visual Code.

When trying to connect to a server that is offline, I wish to sleep a bit and retry connecting.

After calling `socket_connect` twice from the main thread, the board crashes and restarts on the third call. I check the return value from `socket_connect` and it's -1 for failed connects, which I think is the return value of `thrd_suspend(NULL)` in `tcpip_call_input()`?

Unfortunately the crash message is jumbled in the serial monitor, if it actually is a message rather than noise.

I've tried calling `thrd_yield` time and time again around the calls, but I must be missing something?

When the server is online, all works fine. :-)

eerimoq, 2018-01-03:
Hello,

unfortunately, I don't think I have any good advice on how to make it work. It was quite some time since I used the socket module myself, but there is likely a bug or two in it, or in modules it uses, including the espressif code. I usually do some std_printf-debugging to pinpoint problems, but it's a bit tricky in the socket module, as parts of the code is executed from the LwIP thread, where std_printf cannot be used. I'm pretty sure you can use esp_printf (I don't remember the exact name). Would be lovely if you can find a fix for the issue. =)

Glad it works fine when the server is online =)

ChriThor, 2018-01-03:
Thanks for the reply.

Seems to be `PANIC_ASSERTN(out_p->stack_low_magic == THRD_STACK_LOW_MAGIC, ESTACK);` that hits. I suppose that a CURRENT thread cannot be rescheduled and the LwIP-thread is supposed to have made a callback and/or switch to make the main thread suspended or something, but this does not happen on failed connect so it falls apart. I'll see if I pick it up again in the future.

simonlbn, 2018-01-21:
Without knowing for sure stack_low_magic sounds like low water mark of the stack... could it be you are running out of stack space on the threads?

Try to run the debugfs command to see stack usage before second connect and see if that may be an issue?

On a related note, some of the Simba (and possibly underlying SDK code) doesn't handle reconnect etc. gracefully. My current ugly hack which works is to simply reboot my MCU when I need to reconnect.