eerimoq/simba #171: Run disconnects with error from esp32_devkitc

cryarchy, 2018-01-15:
I have noticed a recurrent behaviour (I am not sure if it is expected). I am using the Simba development platform and each time I flash my esp32_devkitc specifying the run command, the client get's disconnected after a while.

For example, running the command (http_client example):
```
make -s BOARD=esp32_devkitc SERIAL_PORT=/dev/ttyUSB0 CDEFS_EXTRA="CONFIG_START_NETWORK_INTERFACE_WIFI_SSID=mywifi CONFIG_START_NETWORK_INTERFACE_WIFI_PASSWORD=myPass" run
```

results in:
```
...
INFO: TIMEOUT = 10

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1880
load:0x40078000,len:3408
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (892) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (892) heap_alloc_caps: At 3FFC45A0 len 0001BA60 (110 KiB): DRAM
I (897) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (907) heap_alloc_caps: At 4009BBB4 len 0000444C (17 KiB): IRAM
I (917) cpu_start: Pro cpu up.
I (922) cpu_start: Single core mode
I (928) cpu_start: Pro cpu start user code
I (1167) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
I (1660) cpu_start: Starting scheduler on PRO CPU.
tcpip_task_hdlxxx : 3ffca214, prio:18,stack:2048
I (1683) wifi: frc2_timer_task_hdl:3ffcbcdc, prio:22, stack:2048
I (1683) wifi: pp_task_hdl : 3ffce528, prio:23, stack:8192
I (1693) wifi: mode : softAP (30:ae:a4:18:19:fd)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
0.030:info:main:default: Mounting the non-volatile memory...
0.030:info:main:default: failed.
0.030:info:main:default: Formatting the non-volatile memory...
0.040:info:main:default: failed.
0.040:info:main:default: Mounting the file system...
0.040:info:main:default: failed.
0.060:info:main:default: Formatting the file system...
1.360:info:main:default: done.
1.360:info:main:default: Mounting the file system...
1.370:info:main:default: done.
Connecting to '216.58.211.142'.

make[1]: *** [rerun] Error 1
make: *** [run] Error 2
```

Similarly, running the following command (blink example):
```
make -s BOARD=esp32_devkitc SERIAL_PORT=/dev/ttyUSB0 run
```
results in:
```
...
INFO: TIMEOUT = 10

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1880
load:0x40078000,len:3408
ho 0 tail 12 room 4
load:0x40080000,len:260
entry 0x40080034
I (438) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (439) heap_alloc_caps: At 3FFB2634 len 0002D9CC (182 KiB): DRAM
I (443) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (453) heap_alloc_caps: At 40090D70 len 0000F290 (60 KiB): IRAM
I (463) cpu_start: Pro cpu up.
I (469) cpu_start: Single core mode
I (475) cpu_start: Pro cpu start user code
I (710) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
I (1190) cpu_start: Starting scheduler on PRO CPU.

make[1]: *** [rerun] Error 1
make: *** [run] Error 2
```
**NOTE:** The error happens after a few seconds of the application running.

Is there a way to circumvent this?

eerimoq, 2018-01-15:
Give `TIMEOUT=3600` on the command line to increase the timeout to 3600 seconds.

cryarchy, 2018-01-16:
Thanks a lot for the pointer. I was unaware of the functionality of the console make target. It worked as expected.

closed 2018-01-16 by cryarchy