eerimoq/simba #175: Analog Write does not work on Arduino Nano

KeelyHill, 2018-05-06:
Running the [Analog Write example](http://simba-os.readthedocs.io/en/latest/examples/analog_write.html) on an Arduino Nano does not work. No voltage is emitted. All PWM tried. Tested with the Arduino IDE and library to check the board (and it works). No error codes are returned -- only zero.

eerimoq, 2018-05-07:
I'll not have time to look into this. Sorry.

KeelyHill, 2018-05-07:
No problem :) .  Thought I should at least report my findings.

alberand, 2018-07-01:
I found the problem.
The **analog_output_pin_init** function initialize the pwm pin. Then, **analog_output_pin_write** configure duty cycle of the PWM module but doesn't set frequency (via setting compare register). The code for configuring compare register is placed at the **pwm_port_set_frequency** function.

The **pwm_start** function sets frequency + duty cycle (by calling **pwm_port_set_frequency**). So, by calling the pwm_start function between initialization and write fix the issue. 

I'm not sure where to place this call in the **analog_output_pin_init** or **analog_output_pin_write** with some handling if freq. is already set. Here is how I fixed it:

```
From 6094363622a48f683fe4dd8f836a89f856d7f744 Mon Sep 17 00:00:00 2001
From: alberand <albershteyn.andrey@gmail.com>
Date: Sun, 1 Jul 2018 14:47:19 +0200
Subject: [PATCH] Fix missing PWM frequency configuration for AVR port

---
 src/drivers/basic/analog_output_pin.c | 10 ++++++----
 1 file changed, 6 insertions(+), 4 deletions(-)

diff --git a/src/drivers/basic/analog_output_pin.c b/src/drivers/basic/analog_output_pin.c
index 5cf7d5ef..ef2e0c10 100644
--- a/src/drivers/basic/analog_output_pin.c
+++ b/src/drivers/basic/analog_output_pin.c
@@ -46,10 +46,12 @@ int analog_output_pin_init(struct analog_output_pin_t *self_p,
         return (-1);
     }
     
-    return (pwm_init(&self_p->pwm,
-                     pwm_p,
-                     pwm_frequency(10000),
-                     pwm_duty_cycle(0)));
+    pwm_init(&self_p->pwm,
+             pwm_p,
+             pwm_frequency(10000),
+             pwm_duty_cycle(0));
+
+    return (pwm_start(&self_p->pwm));
 }
 
 int analog_output_pin_write(struct analog_output_pin_t *self_p,
-- 
2.18.0

```

eerimoq, 2018-07-01:
Create a pull request and I'll have a look.

eerimoq, 2018-07-01:
Good find btw, The PWM module was refactored a few months ago, introducing the pwn_start-function. The analog module was not updated accordingly.

PR #192 (2018-07-01): Fix missing PWM frequency configuration for AVR port
Fixes #175

The analog_output_pin_init function initialize the pwm pin. Then, analog_output_pin_write configure duty cycle of the PWM module but doesn't set frequency (via setting compare register). The code for configuring compare register is placed at the pwm_port_set_frequency function.

The pwm_start function sets frequency + duty cycle (by calling pwm_port_set_frequency). So, by calling the pwm_start function between initialization and write fix the issue.

I'm not sure where to place this call in the analog_output_pin_init or analog_ou [... cut]
closed 2018-07-01 by eerimoq