eerimoq/simba #176: /3pp/esptool/esptool cannot run on raspberry pi - Exec format error

vijaygill, 2018-05-24:
Can python based esptool installed using pip be used instead of bundled esptool? I have esp-open-rtos and Sming working with using the esptool.py. I want to try out simba too.

eerimoq, 2018-05-25:
Give it a try.

vijaygill, 2018-05-25:
@eerimoq I did and it fails on raspberry pi when I run "make BOARD=nodemcu" in the blink example directory. The error is as following:

/bin/bash: /home/pitest/esp8266/sdk/simba/3pp/esptool/esptool: cannot execute binary file: Exec format error
/home/pitest/esp8266/sdk/simba/make/gnu/esp.mk:84: recipe for target 'build/nodemcu/blink.bin' failed
make[1]: *** [build/nodemcu/blink.bin] Error 126
make[1]: Leaving directory '/home/pitest/esp8266/github-stuff/simba/examples/blink'
/home/pitest/esp8266/sdk/simba/make/app.mk:111: recipe for target 'all' failed
make: *** [all] Error 2

eerimoq, 2018-05-25:
3pp/esptool/esptool is a program build for x86_64 (normal 64 bits PC). I'm afraid the Raspberry Pi has an ARM CPU, not x86.

vijaygill, 2018-05-25:
Yes I noticed that. I might fork the project and work on making it work on raspberry pi. From documentation, simba seems quite interesting.

eerimoq, 2018-05-25:
Sounds great!

simonlbn, 2018-05-27:
FWIW, I have been using the official esptool (https://github.com/espressif/esptool) on a Raspberry Pi for a while and it works nicely. I build my Simba based image on AWS EC2 (using normal Simba build framework), and then transfer to a Raspberry Pi that's connected to my dev boards.

I just never got around do doing this "cleanly" - it's just a hacked up wrapper shell script.

closed 2018-06-20 by eerimoq