eerimoq/simba #177: Origin of Simba

Droncz, 2018-05-30:
Hello, Erik,

sorry for bothering you this way - I haven´t been able to find another way to contact you with my question.
I´m relatively new in the IoT world and am working (learning to) with ESP32 chip/devkit. In my yet few little projects, I´ve been using MicroPython which was the easiest way to start for me. Now, as I´m discovering extensive features of the chip, I see it (MP) has its limitations (especially when talking about ESP32). Also, I don´t like very much the fact it runs as a task of FreeRTOS being itself a base of an ESP-IDF framework (if I understood it right).
Thus I´m looking around for other frameworks and have stumbled upon your Simba (thanks to Platform.IO) which looks pretty clean and straightforward. But from your quite nice documentation, I didn´t understand what is it based on - what the final compiled binaries are composed of.
In the documentation you state that: "Simba is an RTOS and build framework". Does it mean you´ve implemented it from scratch or do you build on some other RTOS base?

Excuse me, please, if I'm too curious, I just like to know how things work... ;-)

Best redards,
  Pavel Š.

eerimoq, 2018-05-30:
Hello,

in general Simba is implemented from scratch by me. There are a few modules/functions based on other implementations; for example fat16 and logarithmic functions in the math module. Also, the 3pp-folder contains various third party code used by some modules.

In the ESP8266 and ESP32 ports Simba runs its thread scheduler in a single FreeRTOS thread. Without FreeRTOS the WiFi stack would not work, so I had no choice but keeping FreeRTOS and the Espressif SDK as a base.

Droncz, 2018-05-31:
Thanks for clearing things up.

So, if I got it right, regarding Espressif chips, there´s actually not much difference between your framework and the MicroPython or Pumbaa (except the Python is interpreted which kinda add another layer), when it comes to comparing how close is the final code to the "hardware".
All these frameworks are built upon and around the underlying FreeRTOS and running "only" as a thread of it rather than running on the bare hardware, right?

Thanks again.

eerimoq, 2018-05-31:
Yes, you are right.

Some time ago I saw that nuttx was adding support for ESP32, totally replacing FreeRTOS. Maybe it is finalized and usable.

closed 2018-05-31 by eerimoq