eerimoq/simba #179: Deep sleep on esp8266

toomyem, 2018-06-08:
Is it possible to put ESP8266 into deep sleep using simba os?

eerimoq, 2018-06-09:
There is no power interface in Simba, but maybe the Espressif SDK Simba is running on top of provides this functionality.

simonlbn, 2018-06-09:
I played a bit with this some time ago, but haven't finished it as I haven't really needed it yet. At the time my use case was making reboot reliable, but I did fixed that in another way.

Since it was a while ago, I can't recall all the details. I do seem to recall that I got the sleep to work, but then after wakeup wifi kept being off and I couldn't get it on again.

The Espressif API docs are unfortunately a bit on the terse side, so at times I found it useful to read e.g. Arduino code to understand the Espressif API better.

I will probably get back to this at some point as I want battery powered sensors, but I don't know when.

My code from my "dumping ground" branch that are to hacky or incomplete to try and get merged into Simba itself.

https://github.com/simonlbn/simba/blob/bea1614c6b50e31cd3a92bb222f2132ffab7e2c2/src/drivers/network/esp_wifi/station.c#L120

toomyem, 2018-06-10:
I managed to implement this functionality in simba framework. My wemos d1 mini got to deep sleep and then successfully reconnected to wifi after woke up. I will prepare PR with my changes, but I'm not sure where I should put the code. For now I did it in esp_wifi module, as in your example. But is it a correct place? Please, advice.

eerimoq, 2018-06-10:
I'm not sure myself, maybe create a power module?

toomyem, 2018-06-10:
Seems reasonable. I will try to prepare some code and submit a pull request.

toomyem, 2018-06-11:
I made some initial changes in order to add new power module. Please take a look if you wish.
https://github.com/toomyem/simba/compare/master...toomyem:power-module
Next step is to add support for esp32 as is seems to have this feature too.
Regarding tests, I don't know how deep sleep could be tested in real hardware. I added some basic test, but needs to be extended.

eerimoq, 2018-06-11:
Looks very good. The power module fits nicely in the drivers package since it is hardware dependent. Create a pull request and I'll make an initial review with style comments.

toomyem, 2018-06-11:
It would be nice to also add support for other kinds of sleep, but maybe in another PR.

eerimoq, 2018-06-11:
Yeah, easy to extend the power module later. Let's close this issue as deep sleep has been implemented.

closed 2018-06-11 by eerimoq