eerimoq/simba #180: @section command is not working properly in doxygen comments

toomyem, 2018-06-11:
Most of *.h files begins with license which looks similar to this example taken from assert.h:
```
/**
 * @section License
 *
 * The MIT License (MIT)
(...)

#ifndef __KERNEL_ASSERT_H__
#define __KERNEL_ASSERT_H__

/**
 * Make an assert fatal by negating the error code.
 */
#define FATAL(n) (-n)
```

Unfortunately doxygen somehow wrongly parses this and as a result we have license text combined with the description of the first item in the code. In the example above it is FATAL macro definition.

As a result on documentation pages all first items are not described - the combined license and item's description is not displayed at all.

[image]

eerimoq, 2018-06-11:
How about using this comment instead, to make Doxygen ignore it? It solves the problem with the missing description of the first definition in the file. Is there a better way?

``` c
/*
 * The MIT License (MIT)
 *
...
 *
 * This file is part of the Simba project.
 */
```

toomyem, 2018-06-11:
+1 for this solution.

eerimoq, 2018-06-11:
Fixed in cd26e98c2442ec6d6f865989a15cf3f2cb2ab35a.