eerimoq/simba #185: missing esptool.py

toomyem, 2018-06-13:
File _board.py_ for **esp01** board expects to find _esptool.py_ in _3pp/esptool_ directory, but there is only _esptool_.

```
esptool_path = os.path.join(os.environ["SIMBA_ROOT"],
                                "3pp",
                                "esptool",
                                "esptool.py") 
```

eerimoq, 2018-06-13:
Unfortunately I don't remember why I replaced the python esptool with the compiled one. The commit message is not very descriptive. Looks like other boards are using the compiled tool. I must have forgotten to update the esp01 board.

toomyem, 2018-06-13:
Maybe because for esp01 you are using boot_v1.4.bin and "write_flash" function?

toomyem, 2018-06-13:
What's more, in board.mk you are using scripts for esp12:
```
BOARD_PY = $(SIMBA_ROOT)/src/boards/esp12e/board.py
RUN_PY ?= $(SIMBA_ROOT)/src/boards/esp12e/run.py
```

toomyem, 2018-06-13:
In the "Hardware setup" section of the manual there is picture with description saying that there is esp01 in the test bed. So I thought that it is being tested and running.

Actually I was able to flash sample "hello world" type application but what I get are only boot messages, not a "hello" message I'm expecting. But this is another story which I have to take a closer look.

eerimoq, 2018-06-13:
The test setup is a little dusty. Last time I used it was when the latest release was created, about a year ago. =)

simonlbn, 2018-06-13:
FWIW, I think the better long term fix is to just switch to the official [esptool.py](https://github.com/espressif/esptool). I have been using it for a while since I frequently flash from a Raspberry Pi (IE. ARM) so the compiled esptool from Simba doesn't work.

As a bonus, the official esptool.py compress the image during transfer, so is generally faster than the esptool binary.

This has worked fine on all the ESP8266's I have tried.

eerimoq, 2018-06-13:
Feel free to create a PR replacing the compiled program with the latest Python scripts. I can dust off some of my ESP boards to help out testing.

toomyem, 2018-06-15:
I see you are already using esptool.py in esp-idf module, but it is not in newest version.
What do you think would be better. Just replace esptool in 3pp/esptool with esptool.py taken from Espressiff repository, or add this repository as submodule attached in 3pp/esptool directory?

vijaygill, 2018-06-18:
Coming back after a few days ( got pulled away from projects at home by greater powers ), I see that this issue is being looked after by @toomyem (thanks!). I think this will resolve the issue I raised (and planned to work on a few weeks ago) #176.

toomyem, 2018-06-20:
I guess we can close this issue now.

eerimoq, 2018-06-20:
Thanks for the contribution, always appreciated increasing the quality of the project.

closed 2018-06-20 by eerimoq