eerimoq/simba #188: Missing serial port configuration parameters for shell example

alberand, 2018-06-22:
Hi,
First of all, very cool project

The [shell example](http://simba-os.readthedocs.io/en/latest/library-reference/oam/shell.html)  documentation miss any information about serial port configuration. The [shell library](http://simba-os.readthedocs.io/en/latest/library-reference/oam/shell.html) has link to the system configuration file description, but it also don't provide any information about the default settings. The parameters (e.g. baudrate) can be found in the [default config.h](https://github.com/eerimoq/simba/blob/master/src/config.h)

Therefore, to find this information is quite a challenging process. The shell example contains the config.h file but anyway you need to find out which constants need to be defined. I think there is need to add description of the default serial port configuration to the [shell example](http://simba-os.readthedocs.io/en/latest/library-reference/oam/shell.html).

eerimoq, 2018-06-22:
Hello,

glad you like the project!

Maybe add a Serial Port subsection in the User Guide section in the documentation as the serial port is used in almost all examples and user applications. Then we could add links from the examples to that page.

alberand, 2018-06-24:
Yeah, I think the [User Guide -> Configuration](http://simba-os.readthedocs.io/en/latest/user-guide/configuration.html) is the right place for that (at the bottom of the page). I will make a draft and then PL it.

By the way, what about adding more information to the [Environment setup](http://simba-os.readthedocs.io/en/latest/user-guide/environment-setup.html) about the other available environments (platformio, Arduino IDE). Or better merging the Environment setup with [Build system](http://simba-os.readthedocs.io/en/latest/user-guide/build-system.html) subsection. They complement each other and as they don't contain much information it would be convenient to have it in one place.