eerimoq/simba #193: Use external interrupts with esp01

gacha, 2018-07-02:
Hi, I wanted to use motion sensor with esp01 and tried "exti" module, but method `exti_init` takes only `exti_device_t * dev_p` and I cannot find those pin definitions for esp01. Can I use interrupts with esp01 in some other way?

eerimoq, 2018-07-02:
Hello,

the definitions are [here](https://github.com/eerimoq/simba/blob/master/src/mcus/esp8266/mcu.c#L51-L68). Include `simba.h` to access it. Defines should be added in all ESP8266 board header files. Can you help out doing that?

gacha, 2018-07-03:
Will try to understand how that should be done.

closed 2018-07-04 by eerimoq

gacha, 2018-07-08:
After merge into master this is not working anymore. The last working commit is my f6160d3ff947a3bef5a0b3878d80eef1b6a61e  - and starting with this merge commit 9f1d9e0d50ebfcb9955cfe690 it breaks. After upload the LED blinks very fast and in console I see this repeating:

```
OS SDK ver: 1.3.0(68c9e7b) compiled @ Nov  2 2015 18:53:21
phy ver: 484, pp ver: 9.9

mode : softAP(1a:fe:34:99:a1:5f)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
��@*ר�+�,�lժ�.�hת���.�+�-Hj�NX���*�,�,�,�*��A��/@��*�/%�n�*�����nA,X����*���k�,�*��A��*���(���n�*�����nA,v����*���k�-�*��A.�*���(���n�*Ţ���nA,r␺v�^P�����.�������¹VR���dA%�*�(�(�(�"��@�F %+�(�(����^�@b �j�(U�v@�%p
                                                                     �z��B�X�B��RvP�P�|`rRP��
                                                                                                ��pp_task_hdl : 3fff45f0, prio:13, stack:512
pm_task_hdl : 3fff4ec0, prio:1, stack:255
tcpip_task_hdl : 3fff4370, prio:10,stack:512
idle_task_hdl : 3fff4410,prio:0, stack:255
tim_task_hdl : 3fff67b8, prio:2,stack:256
xPortStartScheduler
frc2_timer_task_hdl:3fff6cc8, prio:12, stack:512
```

My test code is this:

```c
#include "simba.h"

static struct exti_driver_t exti_motion_sensor;
static struct pin_driver_t motion_sensor;
static struct pin_driver_t led;

static void on_motion(void *arg_p) {
  pin_toggle(&led);
}

int main() {
  /* Start the system. */
  sys_start();

  /* LED */
  pin_init(&led, &pin_led_dev, PIN_OUTPUT);
  pin_write(&led, 0);

  /* Motion sensor */
  pin_init(&motion_sensor, &pin_d2_dev, PIN_INPUT_PULL_DOWN);
  exti_init(&exti_motion_sensor, &exti_d2_dev, EXTI_TRIGGER_BOTH_EDGES,
            on_motion, NULL);
  exti_start(&exti_motion_sensor);

  return (0);
}
```

I tested several esp01 boards and different commits and the problem persists. Blink example still works. Maybe someone see some breaking change in this https://github.com/eerimoq/simba/compare/f6160d3ff947a3b...9f1d9e0d50ebfcb9955c diff. Also I don't have any `config.h`.

Let me know if more info is needed.

eerimoq, 2018-07-08:
I'll not have time to troubleshoot this, I'm afraid. Sorry.