eerimoq/simba #200: Upgrade support for esp family

toomyem, 2018-07-11:
Is such functionality planned to be implemented for ESP8266 mcus? I see it is done for ESP32, but for 8266 also should be doable.

vijaygill, 2018-08-25:
Looking at https://github.com/eerimoq/simba/tree/master/src/boards, simba does support esp8266 in various flavours - esp01, esp12e, nodemcu and wemos d1 mini.

toomyem, 2018-08-27:
Nope, upgrade is not supported for ESP8266, take a look at https://github.com/eerimoq/simba/tree/master/src/oam/ports/esp -> where is no upgrade.i implementation for esp.

simonlbn, 2018-08-27:
I looked at it a while ago. The ESP32 SDK provides a clean way of doing this with two firmware images - the ESP8266 SDK does not have this funtionality.

Simba does (as far as I recall) use the Arduino bootloader which does support a way of doing OTA upgrade, but it's not as clean (including that there isn't a fallback if the new image is bad) and requires more custom code. It can probably be done with too much work, but it's not a quick thing and may need a different Simba API than the ESP32 upgrade code.