eerimoq/simba #208: Pro Micro blink example not working

mindforger, 2019-09-08:
Using Windows 10 with Arduino 1.8.9 here, searching for a solution to get multiple COM Ports + HID Keyboard/Mouse/Gamepad

downloaded the Board, selected Simba -> Pro Micro
selected Examples -> Simba -> Blink

upload over a once Arduino Blink Sample

upload finishes, bootloader pops up, vanishes and is replaced with an unknown device due to an identification error and no LED blinking

i can revive the board by repeatedly hitting the reset button (added one for that purpose, bootloader reset can't work without proper serial port) and releasing in just the right moment when trying to flash back the arduino blink sample

so i am pretty sure the code does not work alltogehter (because of the not blinking LED) but i still get some sort of device recognized

VID 000 PID 0003

i also double checked to get a grip of your "pin_device" order (this should seriously be explained somewhere) that it is actually PC0 aka D17 aka RX LED and redid the Arduino Blink Example with it as it uses D13 by default and it does actually blink fine!

eerimoq, 2019-09-08:
I'm sorry, but I don't have time to help you. I hope you figure it out.

mindforger, 2019-09-08:
i guess i've seen something similar using the Raspberry Pi as GAdget, you seem to miss just a few bits of information on the usb descriptor to make it run under windows