eerimoq/simba #210: Unable to build with DHT from PlatformIO

golyalpha, 2020-01-01:
When compiling a project with PlatformIO, targeting Arduino Mega and Arduino Due, it seems that it is unable to find the DHT sensor driver.

Here's the code:
```c
#include "simba.h"
#include "pins.h"
#include "config.h"
#include "events.h"
#include "sensor_threads.h"

#ifndef DHT_SENSOR
// Omitted
#else

static THRD_STACK(dht_sensor_stack, 256);

void *dht_sensor_thread(void *arg_p){
    dht_module_init(); // Mega - W;Due - E:Implicit function declaration

    struct pin_device_t pin_targets[] = DHT;
    int pincount = sizeof(pin_targets)/sizeof(struct pin_device_t);
    struct dht_driver_t sensors[pincount]; // E:Incomplete element type

    for (int i=0;i<pincount;i++){
        dht_init(&sensors[i], &pin_targets[i]); // Mega - W;Due - E:Implicit function declaration
    }

    while (1) {
        int sensor_count=pincount;
        float temperature_acc=0;
        float humidity_acc=0;
        for (int i=0;i<pincount;i++){
            float temp, humidity;
            int res = dht_read(&sensors[i], &temp, &humidity); // Mega - W;Due - E:Implicit function declaration
            if (res != 0){
                std_printf(
                    FSTR("Read failed from DHT sensor %d with %d: %S\n"),
                    i,
                    res,
                    errno_as_string(res) // Mega - W;Due - E:Implicit function declaration
                );
                sensor_count--;
                continue;
            }
            temperature_acc+=temp;
            humidity_acc+=humidity;
        }
        
        float temp = temperature_acc/sensor_count;
        float hmid = humidity_acc/sensor_count;
        if (temp > 30){
            int mask = MAXTEMP_EVENT;
            event_write(&temp_event, &mask, sizeof(mask));
        }
        if (temp < 10){
            int mask = MINTEMP_EVENT;
            event_write(&temp_event, &mask, sizeof(mask));
        }
        if (hmid > 50){
            int mask = MAXHMID_EVENT;
            event_write(&hmid_event, &mask, sizeof(mask));
        }
        if (hmid < 10){
            int mask = MINHMID_EVENT;
            event_write(&hmid_event, &mask, sizeof(mask));
        }
    }
}

#endif

void register_threads(){
#ifndef DHT_SENSOR
    thrd_spawn(temp_sensor_thread, NULL, 2, temp_sensor_stack, sizeof(temp_sensor_stack));
    thrd_spawn(hmid_sensor_thread, NULL, 2, hmid_sensor_stack, sizeof(hmid_sensor_stack));
#else
    thrd_spawn(dht_sensor_thread, NULL, 2, dht_sensor_stack, sizeof(dht_sensor_stack));
#endif
}
```

eerimoq, 2020-01-01:
I'm sorry, but I don't have time to give any support.

golyalpha, 2020-01-01:
I'm more interested in if this is an issue with PlatformIO itself, or not.

golyalpha, 2020-01-01:
Alright, I managed to move ahead. Everyone (Arduino IDE and PlatformIO) are on the latest release... Which does not contain the DHT sensor drivers.