eerimoq/simba #211: Is there missing struct initialization? (http_websocket_client_connect)

RuslanUsmanov, 2020-03-04:
My websocket client connection always return "I/O error" and I don't understand why.

There is a function `int http_websocket_client_connect(struct http_websocket_client_t *self_p)` in  http_websocket_client.c source file

There is a `server_addr` struct declaration:
    `struct inet_addr_t server_addr;`
But there is no `server_addr` initialization or any assignments of values. 
Next there is function `socket_connect` call with a pointer to this struct.
```
    if (socket_connect(&self_p->server.socket, &server_addr) != 0) {
        (void)socket_close(&self_p->server.socket);

        return (-EIO);
    }
```
Is something wrong here? Or am I doing something wrong?

I use platfromio. 
Simba version is 15.0.3

eerimoq, 2020-03-04:
Looks like a bug. Can you try to fix it?

RuslanUsmanov, 2020-03-05:
I added this lines below `struct inet_addr_t server_addr;`. And now there is no more "I/O errors".
```
    if (inet_aton(self_p->server.host_p, &server_addr.ip)  != 0) 
    {
        std_printf(FSTR("Bad ip address %s.\r\n"), self_p->server.host_p);
        return (-1);
    }
    server_addr.port = self_p->server.port;
```

closed 2020-03-05 by RuslanUsmanov