eerimoq/simba #213: Timer timeouts in threads are broken.

RuslanUsmanov, 2020-03-10:
Hello!
I'm trying to use timer timeouts in threads, but the system crashes every time a timeout occurs.
Here is my code:
```
#include "simba.h"

#define TIMEOUT_EVENT    0x1

static THRD_STACK(worker_stack, 256);

static struct event_t event;
static struct timer_t timer;

static void timer_cb(void *arg_p)
{
    uint32_t mask;

    mask = TIMEOUT_EVENT;
    event_write_isr(&event, &mask, sizeof(mask));
}

static void *worker_main(void *arg_p)
{
    uint32_t mask;
    struct time_t timeout;

    std_printf("Thread started\r\n");

    event_init(&event);

    // Initialize and start a periodic timer. 
    timeout.seconds = 1;
    timeout.nanoseconds = 0;
    timer_init(&timer, &timeout, timer_cb, NULL, TIMER_PERIODIC);
    timer_start(&timer);

    while (1) {
        mask = TIMEOUT_EVENT;
        event_read(&event, &mask, sizeof(mask));

        std_printf(FSTR("timeout\r\n"));
    }
    return 0;
}

int main()
{
    sys_start();

    thrd_spawn(worker_main, NULL, 0, worker_stack, sizeof(worker_stack));

    thrd_suspend(NULL);

}
```
Output:
```
pp_task_hdl : 3fff3fe8, prio:13, stack:512
pm_task_hdl : 3fff48b8, prio:1, stack:255
tcpip_task_hdl : 3fff3d68, prio:10,stack:512
idle_task_hdl : 3fff3e08,prio:0, stack:255
tim_task_hdl : 3fff61b0, prio:2,stack:256
xPortStartScheduler
frc2_timer_task_hdl:3fff66c0, prio:12, stack:512

OS SDK ver: 1.3.0(68c9e7b) compiled @ Nov  2 2015 18:53:21
phy ver: 484, pp ver: 9.9

mode : sta(3c:71:bf:31:68:13) + softAP(3e:71:bf:31:68:13)
add if0
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
0.110:info:main:default: Mounting the non-volatile memory...
0.120:info:main:default: failed.
0.120:info:main:default: Formatting the non-volatile memory...
0.130:info:main:default: failed.
0.130:info:main:default: Mounting the file system...
0.180:info:main:default: done.
Thread started
�ht�a|␁��쩓.�dru-@��.�Ѫ  Ko8a ��*�T���   ��h��@��*␓␁
L(A�
��vA,P-�r�␀��N�R�I���/��pp_task_hdl : 3fff3fe8, prio:13, stack:512
pm_task_hdl : 3fff48b8, prio:1, stack:255
tcpip_task_hdl : 3fff3d68, prio:10,stack:512
idle_task_hdl : 3fff3e08,prio:0, stack:255
tim_task_hdl : 3fff61b0, prio:2,stack:256
xPortStartScheduler
frc2_timer_task_hdl:3fff66c0, prio:12, stack:512

OS SDK ver: 1.3.0(68c9e7b) compiled @ Nov  2 2015 18:53:21
phy ver: 484, pp ver: 9.9

mode : sta(3c:71:bf:31:68:13) + softAP(3e:71:bf:31:68:13)
add if0
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
0.110:info:main:default: Mounting the non-volatile memory...
0.120:info:main:default: failed.
0.120:info:main:default: Formatting the non-volatile memory...
0.130:info:main:default: failed.
0.130:info:main:default: Mounting the file system...
0.180:info:main:default: done.
Thread started
�@*r␅nP��␓KM.�d��-@��.�Ѫ        KE�8���*�T��␑���е␑��kA���nz(����nA,P-�r~�P-�R��%��/��pp_task_hdl : 3fff3fe8, prio:13, stack:512
pm_task_hdl : 3fff48b8, prio:1, stack:255
tcpip_task_hdl : 3fff3d68, prio:10,stack:512
idle_task_hdl : 3fff3e08,prio:0, stack:255
tim_task_hdl : 3fff61b0, prio:2,stack:256
xPortStartScheduler
frc2_timer_task_hdl:3fff66c0, prio:12, stack:512

OS SDK ver: 1.3.0(68c9e7b) compiled @ Nov  2 2015 18:53:21
phy ver: 484, pp ver: 9.9

mode : sta(3c:71:bf:31:68:13) + softAP(3e:71:bf:31:68:13)
add if0
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
0.110:info:main:default: Mounting the non-volatile memory...
0.120:info:main:default: failed.
0.120:info:main:default: Formatting the non-volatile memory...
0.130:info:main:default: failed.
0.130:info:main:default: Mounting the file system...
0.180:info:main:default: done.
Thread started
�@*r-nP␙��K�t�dͭ�%+P�jX�Ph ���(P9�␂���)�@��@��*�␁
L(C�
��vA,P'Cr�␀���Эt]�P �
�p
```
Simba version is 15.0.3

eerimoq, 2020-03-11:
Did you try a bigger stack?

RuslanUsmanov, 2020-03-11:
Oh... Thanks. It worked.

closed 2020-03-11 by RuslanUsmanov