eerimoq/simba #218: PIN_INPUT_PULL_UP has been deleted

fletchjp, 2021-07-13:
Hi, I am new to using Simba. I am working on Arduino Uno using the Arduino IDE. I have the hello world and blink examples working. I want to convert an Arduino code I have which uses the PIN_INPUT_PULL_UP option which is in the documentation for Simba version 10 (I found that documentation) but not I find in version 15.

Is there some way to configure a pin for PIN_INPUT_PULL_UP using Simba?

Thanks

John Fletcher

eerimoq, 2021-07-13:
I'm sorry, but I don't have time to help right now. I suggest that you browse the Simba source code to find out what is available. The pin modules and their ports shouldn't be too hard to find in the src/drivers directory.

Erik

fletchjp, 2021-07-13:
Erik

Thank you for your response.

Would you at lease indicate which was the last version to have the pullup code?

John

fletchjp, 2021-07-13:
It seems to me that the problem is that when I am running Simba using the Arduino IDE I am picking up an old version of Simba, with the wrong version of the file pin.h. I don't know how that has come about. I have 15.0.3 installed according to the boards manager and when I click on more_info it tales me to the master branch in GitHub with the version of pin.h which defines PIN_INPUT_PULL_UP. Yet when I compile and run it finds the one without that definition.

eerimoq, 2021-07-14:
It was likely a very long time since the latest Simba release for Arduino. As the interest has been close to zero for Simba on Arduino, and Simba in general, I don't put more time into the project at the moment. Sorry.

fletchjp, 2021-07-14:
I have located the updated version of the files. The file structure for Arduinos is different from the master download and I cannot see how to swap the files in without a lot of work. There must be a way to configure the Arduino installation. I am stopping work on this.